mck1117/wideband #16: Add hardware for onboard voltage sensing.

kjngineering, 2021-07-14:
Currently the controller uses the voltage sensed at the ECU and transmitted via CAN to adjust the heater PID.
Chassis wiring and ground points could mean reasonable differences between the battery voltage at the module and the ECU.

Suggest incorporating an ADC channel and appropriate divider on the module to measure "local" battery voltage for heater compensation.

rusefillc, 2023-02-03:
Another argument for same would be having 042 design self-contained.

rusefillc, 2023-02-03:
https://rusefi.com/forum/viewtopic.php?p=47055#p47055

[image]

rusefillc, 2023-02-03:
Another option https://github.com/mck1117/wideband/issues/194

mck1117, 2023-02-03:
hardware done 6831f5a1ae87e985e1e40e1fd2b6fe2107d40a72

closed 2023-02-04 by mck1117