mck1117/wideband #17: Add switchable RC circuits to support LSU-4.9 and LSU-ADV?

kjngineering, 2021-07-15:
Much like the ST driver it makes sense to have switchable RC circuits to support different sensors. 
Eventually the LSU-ADV will become the standard part.

rusefillc, 2022-04-04:
https://www.bosch-motorsport.com/content/downloads/Raceparts/Resources/pdf/Data%20sheet_69086859_Lambda_Sensor_LSU_ADV/ADV_pre_Turbo.pdf

@kjngineering any part numbers for me to purchase a test LSU-ADV for @dron0gus?

kjngineering, 2022-04-04:
Updated datasheet with Bosch Global PN: https://www.efisolutions.com.au/assets/brochures/135-017.pdf

Post-Turbo: 0 258 027 010
Pre-Turbo: 0 258 027 211

Some Links:
https://www.14point7.com/products/lsu-adv-wideband-o2-sensor
https://www.efisolutions.com.au/bosch-sensor-lsu-adv-wideband-sensor

rusefillc, 2022-04-04:
@kjngineering while those are amazing links i would expect something more OEMish to be cheaper?

either OEM has used ADV somewhere (and we should learn where) or if ADV is just for the fanciest guys there maybe we are not there yet

kjngineering, 2022-04-04:
I did see some OEM cross reference for these but they were out of stock and around the 80EUR level. What is the preferred country of purchase?

rusefillc, 2022-04-04:
USA new/used ebay.com
RU used by model/year

my focus is mostly to make sure that we have a "proper" source

kjngineering, 2022-04-04:
So its apparently the same as Audi/VW/Porsche: 06L906262 (unconfimed)

https://www.ebay.com/itm/334323999253

rusefillc, 2022-04-04:
[image]

kjngineering, 2022-04-04:
> [image]

This is bosch OE 0 258 027 174, according to the product subcategory on buycarparts.co.uk this is a LSUADV sensor.
However, according to the information available on the Audi catalog this is a pre-turbo variant.

The 'best' way to identify a LSU-ADV at the moment is the connector type, they have 5 pins and no internal trim resistor. (this sensor is in line with this information).

mck1117, 2022-04-04:
> this is a pre-turbo variant.

I think the pre-turbo operates the same as the "normal" LSU ADV, but has different thermal behavior

rusefillc, 2022-04-07:
Some info at https://rusefi.com/forum/viewtopic.php?p=44904#p44904