mck1117/wideband #34: Wideband failed to startup after cranking

BeerMoneyMotorsports, 2021-12-30:
Log Below- wbo never showed a reading after engine started.  after a few minutes, i did a firmware write on the wbo while engine running.  45 seconds later, wbo was reading like it should.

https://rusefi.com/online/view.php?log=524

BeerMoneyMotorsports  6:00 PM
@Matthew Kennedy so i have noticed an issue with wbo when long cranking.  i will try to get a log of it and create an issue.
It seem that if you take too long to start or it fails to start, the sensor will not read.  I noticed this while testing on this NB1.

Matthew Kennedy  6:00 PM
hm

BeerMoneyMotorsports  6:00 PM
i have let it run after the car started for a couple minutes
6:00
sensor never shows up

Matthew Kennedy  6:00 PM
loggy log in wbo debug mode

BeerMoneyMotorsports  6:01 PM
i can flash the wbo firmware real quick and then it will heatup and work
6:01
while engine is still running

Matthew Kennedy  6:01 PM
yea
6:01
whatever thermal conditions you're making are probably just barely outside what it can handle

BeerMoneyMotorsports  6:01 PM
ill do a good log tomorrow and toss it in an issue with a reasonable description

Matthew Kennedy  6:01 PM
which means we just need to widen the envelope of "legal" and/or adjust heater tuning a bit (edited) 

BeerMoneyMotorsports  6:02 PM
it seems like its related to the rpm requirement for heating.  does it trip out if jumping between low and high rpm?  or when trig errors occur?

BeerMoneyMotorsports  12:13 PM
@Matthew Kennedy https://rusefi.com/online/view.php?log=524

rusefi.comrusefi.com
rusEFI online
rusEFI tune file sharing site
12:14
wbo never started to read after startup.  after a few minutes, i did a firmware write on the wbo while engine running.  45 seconds later, wbo was reading like it should

BeerMoneyMotorsports  1:36 PM
Which repo do i ticket?

Matthew Kennedy  1:41 PM
Do it in mine
1:41
Probably a bug in the wbo itself

mck1117, 2022-01-10:
likely fixed by 437efb1cc6f5db2aebb5fa5fa3f2d8bebf3716c6

BeerMoneyMotorsports, 2022-01-14:
firmware update but wbo still inop

https://rusefi.com/online/view.php?log=566

BeerMoneyMotorsports, 2022-01-17:
latest firmware (1/17/2022) with no failure.

https://rusefi.com/online/view.php?log=575

closed 2022-10-03 by BeerMoneyMotorsports