mck1117/wideband #44: F103 CAN baud rate

rusefillc, 2022-01-13:
before https://github.com/mck1117/wideband/commit/86034506b07e8c83ea32b3456d0b44037c6cc9da we had

[image]

[image]

rusefillc, 2022-01-13:
after https://github.com/mck1117/wideband/commit/86034506b07e8c83ea32b3456d0b44037c6cc9da we have 
[image]

rusefillc, 2022-01-13:
@blundar great news HW works with 250K version of FW

[image]

mck1117, 2022-01-13:
[image]

rusefillc, 2022-01-13:
I cannot explain it i just report my test results sorry

rusefillc, 2022-01-23:
Just to confirm - is it OK to test with a blue pill with logic analyzer?

[image]

[image]

thirstyone, 2022-01-24:
don't you need some other devices on the bus?

rusefillc, 2022-01-24:
@thirstyone please consider using more words. We are trying to discuss pretty complex issues here and in my opinion short form is not effective. If you are looking to help please consider typing a bit more works to spell things out a bit more explicitly.

@thirstyone stm32 uses TTL levels on RX and TX pins right? Are you saying that without a real transceiver and without loopback I would never see CAN peripheral TTLing anything readable?

mck1117, 2022-01-24:
> without loopback I would never see CAN peripheral TTLing anything readable?

tie TX and RX together, with a 1k pullup to 3.3v, you don't need a CAN tranciever, but you may need a "bus" enough to fake it.

rusefillc, 2022-01-24:
@thirstyone thank you - with real bus I got 8MHz F103 to work! (alligator clips are wired to CANHacker with 120R)

[image]

[image]

rusefillc, 2022-01-24:
and F103 WBO firmware CAN works as well as is on this blue pill setup, I have no idea what was wrong with my previous experiments on real WBO board

[image]

closed 2022-01-24 by rusefillc