mck1117/wideband #168: AUX output

pink-fr3ud, 2022-11-21:
Hi,
Thank you for sharing this.
I was looking at your schematics to design a board with in stock components from JLCPCB assembled
I can't find where the analog output is, as stated in the forum https://rusefi.com/forum/viewtopic.php?f=4&t=1856
In fact, the STM32F042K6T6 doesn't provide a DAC at all...? I see in auxout.cpp you are doing some PWM things, is it how you produce the analog signal?
The STM32G030K6T6 go for as little as $1.1150, is pin to pin compatible, faster, has more memory, DAC ... newer. Maybe a good candidate?

rusefillc, 2022-11-21:
> HI can't find where the analog output is, as stated in the forum https://rusefi.com/forum/viewtopic.php?f=4&t=1856

There are 131 posts in that https://rusefi.com/forum/viewtopic.php?f=4&t=1856 thread please be more specific?

Original hardware design does not have AUX output on purpose, it's only the F103 alternative designs which are trying to add aux output. But those F103 efforts are in a bit of an unclear state poke @dron0gus 

See also #12

pink-fr3ud, 2022-11-21:
Sorry I was refering to #1 post : 
"Support both in-ECU use, as well as standalone. That means SPI or serial comms with an ECU, and CAN or analog for external use."
My bad I didn't see that the F103 was derivated from your design, and indeed he confirmed the DAC working too
I will read a bit more about it and see if I can propose something similar with another mcu

For sure 1.129 Lambda seems poor enough for most tuning needs
Thank you

closed 2022-11-21 by pink-fr3ud

mck1117, 2022-11-21:
Internal and external original design options are both CAN bus