mck1117/wideband #170: f1 dual hard fault on boot

mck1117, 2022-12-02:
[image]

rusefillc, 2022-12-02:
@dron0gus

mck1117, 2022-12-02:
it looks like this might be a gcc11 issue…….stay tuned

dron0gus, 2022-12-02:
I bet on stack overflow. My feeling is that GCC11 produce code that uses more stack.

mck1117, 2022-12-02:
Problem happens on F1, but not F0.

F0 = cortex-m0 aka ARMv6-M
F1 = cortex-m3 aka aARMv7-M

mck1117, 2022-12-02:
|Compiler|F0?|F1?|
|---|---|---|
|GCC 9.3.1|❎|❎|
|GCC 11.3.1|❎|❌|

dron0gus, 2022-12-02:
@mck1117 your master. f1_dual with GD32. GCC11.3.rel1. No HardFault.

```
(gdb) info thread
  Id   Target Id                                                  Frame 
  2    Thread 536873560 (Name: main, State: SLEEPING)             chSchGoSleepS (newstate=8 '\b') at ChibiOS/os/rt/src/chschd.c:558
  3    Thread 536873776 (Name: idle, State: CURRENT)              chSchDoPreemption () at ChibiOS/os/rt/src/chschd.c:765
  4    Thread 536876976 (Name: noname, State: SUSPENDED)          chSchGoSleepS (newstate=3 '\003') at ChibiOS/os/rt/src/chschd.c:558
  5    Thread 536877552 (Name: noname, State: SLEEPING)           chSchGoSleepS (newstate=8 '\b') at ChibiOS/os/rt/src/chschd.c:558
  6    Thread 536878056 (Name: noname, State: SLEEPING)           chSchGoSleepS (newstate=8 '\b') at ChibiOS/os/rt/src/chschd.c:558
  7    Thread 536880360 (Name: noname, State: SLEEPING)           chSchGoSleepS (newstate=8 '\b') at ChibiOS/os/rt/src/chschd.c:558
  8    Thread 536875744 (Name: noname, State: QUEUED)             chSchGoSleepS (newstate=4 '\004') at ChibiOS/os/rt/src/chschd.c:558
  9    Thread 536876456 (Name: noname, State: QUEUED)             chSchGoSleepS (newstate=4 '\004') at ChibiOS/os/rt/src/chschd.c:558
  10   Thread 536879864 (Name: Primary TS Channel, State: QUEUED) chSchGoSleepS (newstate=4 '\004') at ChibiOS/os/rt/src/chschd.c:558
  11   Thread 536878824 (Name: egt, State: SLEEPING)              chSchGoSleepS (newstate=8 '\b') at ChibiOS/os/rt/src/chschd.c:558

```

mck1117, 2022-12-02:
can you attach your `wideband.elf`/`wideband.list`/`wideband.map` that are working?

dron0gus, 2022-12-02:
[wideband.zip](https://github.com/mck1117/wideband/files/10144405/wideband.zip)

mck1117, 2022-12-03:
no crash running your elf on my board

mck1117, 2022-12-03:
double interesting: running the build from github actions does NOT crash, works fine

mck1117, 2022-12-03:
I had a bad compiler install, it turns out if you end up with the wrong `bits/c++config.h`, strange things happen!

closed 2022-12-03 by mck1117