mck1117/wideband #215: No reading for 30 seconds after engine restart

rusefillc, 2023-03-06:
I am running my engine for a while i.e. my sensor is hot.

I stop engine, I start it right away

expected result: i get reading soon
actual result: on dual rev0 it takes 30 seconds to get reading over CAN after restart https://rusefi.com/online/view.php?log=1185

rusefillc, 2023-03-06:
@dron0gus @mck1117 is that a common thing or dual thing?

mck1117, 2023-03-06:
The first start didn't actually get the sensor hot.  The sensor never got in to closed loop the first time (and certainly didn't for very long), and certainly didn't do it for very long. The thermal conductivity out of the sensing element is significant - the element itself is relatively light for fast warmup, which means it cools off quick too.

Everything here looks normal to me.

[image]

mck1117, 2023-03-06:
Is that controller using internal vbatt measurement (which might be bogus?) or listening to vbatt from rusEFI?

rusefillc, 2023-03-06:
this same sensor was running 29 of the previous 30 minutes, i would assume it to be pretty warm

no idea on vbatt, 2chan next rev is a week away from NJ

mi-hol, 2023-06-14:
@rusefillc just reading thru the comments and was wondering if "the ball" (= answer last question) is not "laying in your court"?

rusefillc, 2023-07-15:
closing due concerns about dual-dev0 vbatt implementation

closed 2023-07-15 by rusefillc