mck1117/wideband #296: how fast do we expect things to start?

rusefillc, 2023-11-22:
hellen 042
first release

not first start today - exhaust and sensor are not completely cold

yet it seems to take ~45 seconds to see WBO reasing
(also all that time fault code is just 0 maybe I shall migrate to fresh binary)

[image]
[2023-11-22_14.52.35 with sensor.mlg.zip](https://github.com/mck1117/wideband/files/13443961/2023-11-22_14.52.35.with.sensor.mlg.zip)

[image]

rusefillc, 2023-11-23:
Now with today snapshot https://github.com/rusefi/wideband/releases/tag/20231122-2 60 seconds to report AFR.

is it a coincidence that I had to touch throttle to get reading? Can wideband detect "too lean" and "too rich" and report those as separate states?

[2023-11-22_21.56.09 mck master.mlg.zip](https://github.com/mck1117/wideband/files/13445713/2023-11-22_21.56.09.mck.master.mlg.zip)

[image]

mck1117, 2023-11-23:
@rusefillc give it another try with current master

mck1117, 2023-11-23:
> is it a coincidence that I had to touch throttle to get reading?

nope, that's just a coincidence.

> Can wideband detect "too lean" and "too rich" and report those as separate states?

It could detect that the pump drive is pinned against the stop (and so the nernst voltage isn't being controlled at ~0.45v), we could at least clear the valid bit when that happens.

rusefillc, 2023-11-23:
latest artifact is silent on the CANbus. just reverted to the not-hanging version and CANbus has returned

will now try latest one more time

rusefillc, 2023-11-23:
progress! we now get AFR in about 20 seconds from engine start

[image]
[2023-11-22_22.55.22 ver3.mlg.zip](https://github.com/mck1117/wideband/files/13445907/2023-11-22_22.55.22.ver3.mlg.zip)

mck1117, 2023-11-23:
> clear the valid bit when that happens

this is detected and transmitted in ea46db500f6f59fcefe6273ae104fb3fde2bce23

closed 2023-11-23 by mck1117