mck1117/wideband #299: brand new sensor failed to heat

rusefillc, 2023-11-25:
042 hellen module
june release

[wbo-stars.zip](https://github.com/mck1117/wideband/files/13464705/wbo-stars.zip)

mck1117, 2023-11-26:
[image]

based on the 0.3v jump in battery voltage when the heater kicks off, I'm going to blame weak wiring/psu here

rusefillc, 2023-11-26:
That's on a real bike with a small battery. WBO located on same PCB as main connector hooked to OEM harness.

rusefillc, 2023-11-27:
here's a new log with heater+ taken from a better spot - VBatt jump is much smaller, still failed to warm up

[WBO2 LOG with battery power to heater +.zip](https://github.com/mck1117/wideband/files/13468564/WBO2.LOG.with.battery.power.to.heater.%2B.zip)

[image]

mck1117, 2023-11-27:
try it with the new bin that heats more aggressively

rusefillc, 2023-12-05:
things are better with 2111 pre-release

[DataLogs12-5_LF1.zip](https://github.com/mck1117/wideband/files/13573435/DataLogs12-5_LF1.zip)

new issue https://github.com/rusefi/rusefi/issues/5773

closed 2025-01-28 by mck1117