mck1117/wideband #300: Modifying code for the F0 board

ChubbyBunny1, 2023-11-26:
Can anyone give some directions on how to modify and upload the code to the F0 board, what IDE to use, workflow. I have an ST-LINK V2 and CubeIDE installed  

Thank you

mck1117, 2023-12-06:
Hi, what changes are you trying to make? If it's something that would be useful for others too, I'd be happy to implement them and get them in to official builds.

ChubbyBunny1, 2023-12-06:
I am trying to make an analog output. I modified the PCB and added an opamp with RC filter at PB1 (TIM3_CH1) in order have outpt 0-5V analog. Since timer 3 is set up at high frequency it should be easy to add an extra PWM out right?

rusefillc, 2023-12-06:
> I am trying to make an analog output. 
have you seen F103 boards doing this with a DAC?

ChubbyBunny1, 2023-12-06:
> > I am trying to make an analog output.
> 
> have you seen F103 boards doing this with a DAC?

Yes ofcource but i want to make this board work with analog output too.

closed 2025-01-28 by mck1117