mck1117/wideband #305: DBC file broken

dynfer, 2023-12-12:
The DBC file when viewing its source or via a tool displays an bogus frame ID.

[image]
[image]

https://github.com/mck1117/wideband/blob/master/wideband-can.dbc

rusefillc, 2023-12-12:
@dynfer I would assume that vector CANdb++ is _the_ official too? Is not that tool free?

DBC is a proprietary standard by vector is not it?

mck1117, 2023-12-12:
I created it and it opens just fine in Kvaser Database Editor:

[image]

dynfer, 2023-12-12:
@rusefillc to my knowledge its the official however the program is not free to use hence the other tools I've used to view it.
If it is proprietary I cannot tell.

@mck1117 I assume then this adress is correct and not an error?

mck1117, 2023-12-12:
That address is correct, yes.

dynfer, 2023-12-12:
All clear thanks

closed 2023-12-12 by dynfer