mck1117/wideband #307: Weird behaviour of hand soldered F0 board runing latest firmware

dynfer, 2024-01-20:
Hello,

I have two f0 boards which have two different behaviours, both running November firmware from mck1137/wideband.

1. Nerst voltage fluctuates depening on the applied oxygen (sensor over candle or lighter) afr sometimes jumps if nerst crossed over 450mV
2. Nerst voltage stays fixed at around 450mV, pump current changes depending on applied oxygen, AFR stays pretty much fixed at around 15.45

I've attached logs from both boards 2024-01-18_18.02.05.zip

Both boards are hand assembled so there might an issue with soldering??

//EDIT
I've verified all solder joints and now both units behave like in the log below, the spikes are nerst V and pump DC are from me applying propane to the sensor 2024-01-18_21.41.24.zip
[2024-01-18_18.02.05.zip](https://github.com/mck1117/wideband/files/13998219/2024-01-18_18.02.05.zip)
[2024-01-18_21.41.24.zip](https://github.com/mck1117/wideband/files/13998221/2024-01-18_21.41.24.zip)

rusefillc, 2024-01-20:
@dynfer please describe your power supply

@dynfer maybe add a photo of the overall setup?

dynfer, 2024-01-20:
> @dynfer please describe your power supply
> 
> @dynfer maybe add a photo of the overall setup?

For power I've used a AC/DC 12V 1A power brick and 12v from PC power supply both shared the behaviour

Only thing that would differ from a standard setup that I have soldered wires directly to the pins of the 12pin connector on the carrier board instead of using the connector itself.