mck1117/wideband #313: Old parameters in uart.cpp and linking error

Theo-Point, 2024-04-06:
Hello,
I'm trying to use the serial debug functionnality on a custom standalone F0 board in order to debug a non-trivial heater state error. (over heated on closed loop after 5s, I tried the no heater state branch).

So i defined a DEBUG_SERIAL_PORT according to other boards config but i get few errors :

- First one was the ChibiOS serial driver not being included. But by including the correct file it's all good.

- The second one was that after a certain commit (i looked it up but i can not find it anymore) you changed the `GetSampler(ch)` function and removed the parameter for the channel but it's still there in this optionnal part of uart.cpp
So in the uart thread, `GetSampler(ch).GetNernstDc(ch)` should be `GetSampler().GetNernstDc(ch)` etc...
The problem is the same for `(GetSampler(ch).GetNernstAc(ch)`, `GetSampler(ch).GetSensorInternalResistance(ch)`, `GetSampler(ch).GetSensorTemperature(ch)` and `GetSampler(ch).GetPumpNominalCurrent(ch)`.

After that change it compiles but when linking an error is thrown telling me that according to the linker script, the stm32 doesn't have enough RAM. I can't give you the exact error because my stm32 fried or my st-link did. But it was too much by around 600 bytes. (I tried to remove max3185x.cpp from the project but i just reliased that it was not compiled due to no EGT channels)

Does anyone had this error when trying to use the serial debug functionality on f0 boards ?
Should I open a pull request to modify uart.cpp ?

Thanks for all you work ! 👍

dron0gus, 2024-04-06:
I think no one try serial debug output on f0.
Easiest way to get debug is to connect f0 module to some RusEFI ecu and write log. Wbo emits debug can message, rusefi includes ot into log.
F0 us limited in resources, you can try disabling bootloader to free some flash. But if you are out of RAM this will not help.

dron0gus, 2024-04-06:
Also double check that your Ip current pump is working and correctly connected. Without feeding pump temperature measurement can go wrong.

Theo-Point, 2024-04-06:
Thanks for your response !

I ordered new components for now so I wont be able to try for now, but i will give it a go.

dron0gus, 2024-04-06:
Also you can try this test branch https://github.com/dron0gus/wideband/tree/underheat-issue . Issue is still in debug, but looks similar to yours.

mck1117, 2024-04-06:
> Also you can try this test branch https://github.com/dron0gus/wideband/tree/underheat-issue . Issue is still in debug, but looks similar to yours.

PR?

rusefillc, 2024-04-06:
> PR?

could be blocked by https://github.com/mck1117/wideband/pull/294

dron0gus, 2024-04-07:
@mck1117  https://github.com/mck1117/wideband/pull/314 . But needs more testing.

closed 2025-01-24 by mck1117