mck1117/wideband #318: No CANBUS reading from Wideband to Rusefi ECU

NickZ1969, 2024-05-16:
No CANBUS reading from Wideband to Rusefi ECU. 
Using original release wideband firmware canbus wideband values are received fine but sensor eventually times out on goes into error.
Using second release no sensor timeout but data is displayed as 0.0 and does not change on rusefi.
using latest firmware from this github data is either 29 or 0.0 and nothing in between on rusefi.

I was using rusEFI master.2024.04.18.f407-discovery.137850894 and now updated to latest Rusefi firmware.

replacing with AEM canbus wideband readings are as expected.

mck1117, 2024-05-16:
What controller? Do you have a log from the ECU of warmup/error? Does the wideband show any error blink code? Does this sensor work with other controllers? Does this controller work with other sensors?

NickZ1969, 2024-05-16:
the wideband controller is this controller on the adapter board.
Im sorry I don't have a log of the wideband, how do I log the wideband controller?
with the latest firmware the controller warms the sensor, green LED goes into slow bling, the heater led shows signs of it regulating the heating and all seems to be working as it should but no data over the canbus. No blue led blink errors. the controller doesn't go into error.
yes the sensor/s works perfect when I switch to my AEM controller and ive driven 100s of km on the AEM without errors in the same car.
I have not had the controller work on any other sensors with the later firmware, The first release works fine for about 5 minutes before going into error on all sensors ive tried. A ignition power off and system will work again for a couple of minutes.

mck1117, 2024-05-16:
> Im sorry I don't have a log of the wideband, how do I log the wideband controller?

just grab an ECU log, the ECU logs some parameters from the wideband.

NickZ1969, 2024-05-16:
ok i will make a log later in the day and send it.

regards
Nick

NickZ1969, 2024-05-18:
[2024-05-18_14.45.26.mlg.txt](https://github.com/mck1117/wideband/files/15360434/2024-05-18_14.45.26.mlg.txt)
here is a log from today

VFRenjoyer, 2024-05-19:
I may be having a similar problem on my board with two wideband controllers, the relevant logs are at 
https://rusefi.com/online/view.php?log=1396
https://rusefi.com/online/view.php?log=1397

ElDominio, 2024-05-21:
I also have tune and logs: https://rusefi.com/online/view.php?log=1395

NickZ1969, 2024-06-10:
Any updates on this yet?

NickZ1969, 2024-06-10:
Issue was me. pins 1 and 2 where swapped (sensor pins).

mck1117, 2024-06-16:
So this is fixed, and was just a wiring issue?

NickZ1969, 2024-06-16:
My issue yes, was wiring.

closed 2024-06-17 by mck1117

ElDominio, 2024-06-21:
Mine wasn't. Changed to an OG AEM Wideband and still doing the same thing

mck1117, 2024-06-26:
> Mine wasn't. Changed to an OG AEM Wideband and still doing the same thing

So also not an issue with this controller.

ElDominio, 2024-06-26:
*included 4.9 with an xseries uego

mck1117, 2024-06-26:
> *included 4.9 sensor with an xseries uego

please use more words or I can't help you

ElDominio, 2024-06-26:
I changed the sensor i was using to the included one with an AEM X-Series UEGO, and it still does the same thing.

I removed my controller and used the X Series controller, and it reads fine.

I will upload my schematics tonight

ElDominio, 2024-07-31:
I changed to yet another and it was fine??!? I have two on my 3000GT and it works; maybe it is the car it's on? some electrical noise or something?