mck1117/wideband #322: Potential thread stack overflow in function `AuxOutThread`

mirusu400, 2024-11-11:
### Summary

There are potential thread stack overflow in thread function `AuxOutThread`

### Details

https://github.com/mck1117/wideband/blob/4e3e4d75d4073edae20b8eb1fbebbb514c487e75/firmware/auxout.cpp#L185-L187

In this line, `AuxOutThread` Allows size of `waAuxOutThread`, which is 256 bytes.

However, after manually checking stack, it might have **264** Bytes.

**Steps to reproduce**

For this step, I just use `rusefi` git repo for reproduce.

0. Build rusefi once nomally (environment setup, etc..)

1. Add this flags in `firmware/Makefile`

```diff
...
# Compiler options here.
ifeq ($(USE_OPT),)
  USE_OPT = -O2 -ggdb -fomit-frame-pointer -falign-functions=16 -fsingle-precision-constant
endif

++ USE_OPT += -fstack-usage
...

# Remove LTO for getting stack size dump
++ # Enable this if you want link time optimizations (LTO).
++ # ifeq ($(USE_LTO),)
++ #   USE_LTO = yes
++ # endif
```

2. build again.

Now we can get stack usage file (*.su) for each source file, So we can manually check stack size of each function.

### PoC
**In case of `AuxOutThread`**:
```
AuxOutThread (_Z12AuxOutThreadPv) 104 size
chThdSleep (chThdSleep) 16 size
chThdSleepS (chThdSleepS) 16 size
chSchGoSleepTimeoutS (chSchGoSleepTimeoutS) 40 size
chVTDoSetI (chVTDoSetI) 48 size
port_timer_start_alarm (port_timer_start_alarm) 16 size
stStartAlarm (stStartAlarm) 16 size
st_lld_start_alarm (st_lld_start_alarm) 8 size

=> SUM: 264 bytes
```

So, there are potentially occur stack overflow in `AuxOutThread` Thread function.

I've check all the given controllers and it seems that only `f1_dual` and `f1_rev2` controller have this vulnerability.

mck1117, 2025-03-04:
@mirusu400 what tooling did you use to combine the stack usage and call graph?

> `# Remove LTO for getting stack size dump`

This alters the result - LTO inlines lots of stuff and even removes parameters from functions, so the stack usage is typically strictly less when LTO is enabled.

mck1117, 2025-03-04:
also, looks like `-fstack-usage` still works with LTO on:

[image]