mck1117/wideband #323: Fix build for F1 boards

mck1117, 2025-01-24:
Something changed in newer ChibiOS, maybe F1 boards need tweaking to match.

mck1117, 2025-01-24:
@dron0gus @rusefillc interest? I've disable their builds for now on CI 40d264b2f75594527e3e9825ee3856ea6f119d50

dron0gus, 2025-01-27:
ChibiOS needs patching.
STM32_DAC_HAS_MCR should be false for STM32F1xx... And many others...

dron0gus, 2025-01-27:
https://github.com/mck1117/wideband/pull/324
and
https://github.com/dron0gus/ChibiOS/commit/88cef96dd5f86fc5a73e10fc5f3d13ab6030b9b2

mck1117, 2025-01-28:
> ChibiOS needs patching. STM32_DAC_HAS_MCR should be false for STM32F1xx... And many others...

Seems like a good one to upstream to chibios

dron0gus, 2025-01-28:
> Seems like a good one to upstream to chibios

I'll do.

mck1117, 2025-05-28:
> I'll do.

Any news here?

mck1117, 2025-06-20:
looks like as of 1495e705979f87016fbfa484c36777f1f98ff30a, F1 boards are building again