mck1117/wideband #325: Heater duty oscillating

dron0gus, 2025-01-27:
[image]
Both channels on F1 dual Rev1. Not observed on my master.

mck1117, 2025-01-27:
> Not observed on my master.

have you changed heater PID parameters?

dron0gus, 2025-01-27:
> have you changed heater PID parameters?

Nope. PID parameters are stock.

dron0gus, 2025-01-27:
Ok, seems you lost conversion from mS to S in https://github.com/mck1117/wideband/commit/7954cbd5a441a859e804c328519e33b7654a4c1d .

[image]

rusefillc, 2025-01-27:
@mck1117 @dron0gus any chance for a defect if this turns to be a legit new defect?

dron0gus, 2025-01-27:
https://github.com/mck1117/wideband/pull/326

dron0gus, 2025-01-27:
> Ok, seems you lost conversion from mS to S in [7954cbd](https://github.com/mck1117/wideband/commit/7954cbd5a441a859e804c328519e33b7654a4c1d) .
> 
> [image]

This is a "good" oscillations. After fix. This is damping oscillations. Not sure how fast we expect it to become straight line, but look much better.

mck1117, 2025-01-27:
fixed in 2c1f1acd9f6e5b3b8da9d36d51acf0b44f415393

closed 2025-01-27 by mck1117